ARMmbed/ble-nrf51822 #5: linking esb_gcc.a (nrf51822 enhanced shock burst) with mbed

freber, 2015-01-14:
Gentlemen,

Love your work and trying to get nrf51822 esb (enhanced shock burst) to work with mbed. When linking gcc_esb.a (the esb library) there is a conflict that I would like to get help with.

```
Fredrik:Build freber$ make
Scanning dependencies of target MY_APP.elf
[  1%] Building C object CMakeFiles/MY_APP.elf.dir/mbed-src/libraries/mbed/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib/app_common/app_timer.c.o
In file included from /Users/freber/dev/myApp/mbed-src/libraries/mbed/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib/app_common/app_timer.c:19:0:
/Users/freber/dev/myApp/nRF51822/nordic/nrf-sdk/nrf_delay.h:48:1: warning: 'inline' is not at beginning of declaration [-Wold-style-declaration]
 static void __INLINE nrf_delay_us(uint32_t volatile number_of_us)
 ^
Linking CXX executable MY_APP.elf
../nRF51822/nordic/nrf-sdk/Lib/esb/gcc/esb_gcc.a(nrf_notification_manager.o): In function `SWI0_IRQHandler':
nrf_notification_manager.c:(.text+0xac): multiple definition of `SWI0_IRQHandler'
CMakeFiles/MY_APP.elf.dir/mbed-src/libraries/mbed/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib/app_common/app_timer.c.o:app_timer.c:(.text.SWI0_IRQHandler+0x0): first defined here
../nRF51822/nordic/nrf-sdk/Lib/esb/gcc/esb_gcc.a(nrf_flywheel.o): In function `TIMER2_IRQHandler':
nrf_flywheel.c:(.text+0x1ec): multiple definition of `TIMER2_IRQHandler'
CMakeFiles/MY_APP.elf.dir/mbed-src/libraries/mbed/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/pwmout_api.c.o:pwmout_api.c:(.text.TIMER2_IRQHandler+0x0): first defined here
collect2: error: ld returned 1 exit status
make[2]: *** [MY_APP.elf] Error 1
make[1]: *** [CMakeFiles/MY_APP.elf.dir/all] Error 2
make: *** [all] Error 2
```

0xc0170, 2015-01-14:
please don't duplicate issues. You already submitted this issue to mbedmicro repository. I answered it there, this is related to HAL, so I'll close this one. Thx for understanding.

0xc0170, 2015-01-14:
I noticed you got there 2 errors, SWI0_IRQHandler is another one. So if this one is connected to this repo, then please reopen it. Provide more details like what you are reporting, that there are multiple definitions and can also reference files on github, will help to find the root cause.

closed 2015-01-14 by 0xc0170

rgrover, 2015-01-15:
regarding the multiple definitions for SWI0_IRQHandler, feel free to remove TARGET_MCU_NRF51822/pwmout_api.c from your build tree.

freber, 2015-01-15:
Thanks for all your input both @0xc0170 and @rgrover. Sorry for double posting, but I wasn't sure where it was belonging. @rgrover: is pwmout_api.c not platform specific?

Best regards,
Fredrik

rgrover, 2015-01-15:
pwmout_api.c is platform specific. But if you aren't using pwm in your application it should be safe to drop.