ARMmbed/ble-nrf51822 #10: bring s110 support back

bearsh, 2015-05-18:
I fully understand mbed's decision to concentrate on 4.1 and 4.2 features and have things like
> Support for multiple concurrent connections to the peripheral, and being able to act as a peripheral and central at the same time.

but unfortunately the ram requirements tend to rise with each new release. That's primarily not your fault as newer softdevices need more ram, especially the s130.

Using the 130 softdivice on a 16K chip, only about 6k are available for the application. If the user has the choice between s130 and s110 (and only need the peripheral functionality) he has at least 8k of ram for his application.

So what I propose is, if possible with reasonable effort,to support both softdevices, the s110 and the s130 (maybe even older s110s which need even less ram). Atm it seems to be possible to still use the s110 with the latest revision of this repo.

rgrover, 2015-05-19:
Your suggestion has merit.
The selection of the softdevice is tied into the definition of the Nordic platform; it is a part of the mbed-SDK. Implementing your suggestion would require us to maintain a 'lite' version of the nRF51822 platform with reduced functionality. That would be an added maintenance load. We'd also have to choose a 'lite' softdevice, and partition the BLE_API into a 'lite' subset. Softdevices go back to many older versions. The S110 softdevice is presently at v8, but goes back to v4. The choice of the older lite-softdevice would be arbitrary, and can't be justified only on the basis of size. If we start going down this route, there would emerge other desirable ways to pick the soft-device based on what's needed. And this will be all too Nordic centric.

We are hoping to extend the reach of the BLE_API to other platforms where SRAM may not be a severe restriction. Nordic may also offer nRFs with larger SRAM before too long. And with the S130 it is also possible to reduce the amount of memory taken up by the GATT Server if that's any use--check the documentation that comes with S130.

Given these reasons, the mbed team won't prioritize what you've suggested. Perhaps Nordic can take up the maintenance of their softdevice and offer lite versions for mbed.

closed 2015-05-19 by rgrover

bearsh, 2015-08-27:
support for s110 is comming back:
https://developer.mbed.org/forum/team-63-Bluetooth-Low-Energy-community/topic/17027/?page=1

thanks guys!

zskdan, 2015-08-27:
Goood !! Thank very well !!!