ARMmbed/ble-nrf51822 #11: Debug builds fail due to missing bsp.h

Timmmm, 2015-05-29:
In app_error.c there are the following lines:

    #ifdef DEBUG
    #include "bsp.h"

If you're doing a debug build (offline; I'm not sure the online compiler supports it) then DEBUG is defined and compilation fails because there is no bsp.h.

rgrover, 2015-05-29:
Will the following diff work for you? If this is acceptable, I'll include it in the next release.

```
--- a/nordic-sdk/components/libraries/util/app_error.c
+++ b/nordic-sdk/components/libraries/util/app_error.c
@@ -24,7 +24,7 @@
 #include "compiler_abstraction.h"
 #include "nordic_common.h"
 #ifdef DEBUG
-#include "bsp.h"
+// #include "bsp.h"^M
 
 /* global error variables - in order to prevent removal by optimizers */
 uint32_t m_error_code;
@@ -52,11 +52,11 @@ __WEAK void app_error_handler(uint32_t error_code, uint32_t line_num, const uint
 #ifndef DEBUG
     NVIC_SystemReset();
 #else
-    
-#ifdef BSP_DEFINES_ONLY 
+^M
+#ifdef BSP_DEFINES_ONLY^M
     LEDS_ON(LEDS_MASK);
 #else
-    UNUSED_VARIABLE(bsp_indication_set(BSP_INDICATE_FATAL_ERROR));
+    // UNUSED_VARIABLE(bsp_indication_set(BSP_INDICATE_FATAL_ERROR));^M
     // This call can be used for debug purposes during application development.
     // @note CAUTION: Activating this code will write the stack to flash on an error.
     //                This function should NOT be used in a final product.
```

Timmmm, 2015-06-01:
Yep that works. :-)

commit c28d0eea (2015-08-11): fix #11. Remove include for missing header.
closed 2015-08-11 by rgrover