ARMmbed/ble-nrf51822 #16: Clash With Definition And Enum Naming

jcady92, 2015-06-26:
See: mbedmicro/BLE_API#46

jcady92, 2015-06-28:
I just discovered there's also a file naming clash with the BLE API.  Trying to include "BLE.h" leads to problems because there's also a BLE.h in nrf51822.

rgrover, 2015-06-29:
this is a duplicate of https://github.com/mbedmicro/BLE_API/issues/46
the problem needs to be fixed in nRF51822, not in BLE

rgrover, 2015-06-29:
regarding duplicate BLE.h: the include of BLE.h from BLE_API should be done like:
```
#include "ble/BLE.h"
```

jcady92, 2015-06-29:
I did find one more problem.  There's another naming clash in UARTService.h.

Enum here:

https://github.com/mbedmicro/BLE_API/blob/master/ble/services/UARTService.h#L44

Define here:

https://github.com/mbedmicro/nRF51822/blob/develop/nordic-sdk/components/softdevice/s130/include/ble_gatt.h#L50

Same problem, compiler complains of expected identifier.

commit 3d87035e (2015-06-29): fix #16: hide duplicate symbol definitions in ble_types.h
commit 2ebe5108 (2015-06-30): Fix #16: move GATT_MTU_SIZE_DEFAULT into blecommon.h
Note: there's a corresponding change to mbedmicro/BLE_API
closed 2015-07-02 by rgrover