ARMmbed/ble-nrf51822 #18: service discovery should be able to match against long UUIDs

rgrover, 2015-06-29:
Currently we've tested it only for short UUIDs.

rainierwolfcastle, 2015-10-15:
http://jira-dev0.cambridge.arm.com/browse/IOTSFW-853

rainierwolfcastle, 2015-10-16:
ARM Internal Ref: IOTSFW-1029

jrobeson, 2015-10-22:
added to the nordic stack where?

rgrover, 2015-10-22:
The nordic stack likes to deal with UUIDs using a short internal representation. https://github.com/ARMmbed/ble-nrf51822/blob/master/source/btle/custom/custom_helper.cpp#L82
When a long UUID is discovered during service discovery, it is represented as an invalid type (or a user-defined type; I forget which). The long UUID needs to be fetched explicitly and added to the ble stack before it can be represented again using the short form. https://github.com/ARMmbed/ble-nrf51822/blob/master/source/btle/custom/custom_helper.cpp#L82

pan-, 2016-07-15:
This is fixed with latest versions of nrf51822.
It even discover 128 bits UUID descriptors.

closed 2016-07-15 by pan-