ARMmbed/ble-nrf51822 #30: clearScanResponse()

swatissehgal, 2015-07-29:
I use accumulateScanResponse to send scan Response, but later I want to stop sending the scan Response on a certain condition. I tried using clearScanResponse() for this but the observer device still receives scan response (same as the previously received scan response) if active scanning is enabled on it.

The clearScanResponse() function should allow the advertising device to have the authority of not sending any scan response if it wants.

rgrover, 2015-07-30:
clearScanResponse() clears the scan-response payload, but setting a null scan-response results in a no-op for the nordic stack. This we need to do something about this.

rgrover, 2015-08-07:
will be handled by https://github.com/ARMmbed/BLE_API/issues/64

rgrover, 2015-08-11:
closing this because I believe this has been fixed. Please re-open if you disagree.

closed 2015-08-11 by rgrover