ARMmbed/ble-nrf51822 #35: nrf51822 hangs after calling sd_flash_page_erase()

leibin2014, 2015-08-07:
I'm using nrf51822 for products of our company. We use mbed + nRF51822 driver + BLE_API for the BLE peripheral devices. The softdevice is  s130 ver 1.0.0.

I create an application with the linkscript as below, the secured BLE works fine.
MEMORY
{
FLASH (rx) : ORIGIN = 0x0001C000, LENGTH = 0x24000
RAM (rwx) : ORIGIN = 0x20002800, LENGTH = 0x5800
}

Currently I have a use case that boot this application with a short separate boot code.
I create the boot code as below:
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
#define APPLICATION_ADDRESS 0x00023800
typedef void (*pFunction)(void);
pFunction JumpToApplication;
uint32_t JumpAddress;
void main()
{
 /* Jump to user application */
JumpAddress = *(__IO uint32_t*) (APPLICATION_ADDRESS + 4);
JumpToApplication = (pFunction) JumpAddress; /* Initialize user application's Stack Pointer */ 
__set_MSP(*(__IO uint32_t*) APPLICATION_ADDRESS);
JumpToApplication();
}
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
The linkscript for the boot code is:
MEMORY
{
FLASH (rx) : ORIGIN = 0x0001C000, LENGTH = 0x7800
RAM (rwx) : ORIGIN = 0x20002800, LENGTH = 0x5800
}

And change the linkscript for the application as below:
MEMORY
{
FLASH (rx) : ORIGIN = 0x00023800, LENGTH = 0x1C800
RAM (rwx) : ORIGIN = 0x20002800, LENGTH = 0x5800
}

This means that I put the boot code at 0x0001C000, the application at 0x00023800. Then the after startup the nrf51822 will jump from softdevice code to the bootcode, and then jump to the application code.

I found the application can be executed correctly, but after calling initializeSecurity() the cpu gets stuck. After debugging I found the problem happened after calling sd_flash_page_erase() in pstorage_init(). But the return value for  sd_flash_page_erase() is NRF_SUCCESS.

I tried to replace sd_flash_page_erase() with ble_flash_page_erase(), the problem is gone.
And this problem doesn't exist when start the same application without this boot code (start the application by softdevice directly at 0x0001C000).

Any idea about this?

leibin2014, 2015-08-07:
After I adding sd_softdevice_vector_table_base_set(0x00023800) in function nRF51822n::init(void) of the application, everything works fine. Looks like this function should be called before jump to the application. So the main function  of boot code should be changed. Could you help to review this main function of boot code to check if there is any other thing should be do in it?

void main()

{

 /* Jump to user application */

JumpAddress = *(__IO uint32_t*) (APPLICATION_ADDRESS + 4);

JumpToApplication = (pFunction) JumpAddress;

/* Initialize user application's Stack Pointer */

__set_MSP(*(__IO uint32_t*) APPLICATION_ADDRESS);

JumpToApplication();

}

rgrover, 2015-08-07:
Hi Bin,

Refer to https://github.com/ARMmbed/dfu-bootloader/blob/master/bootloader.c#L304. It will show how the Nordic bootloader jump to the main application. Your code snippet (above) looks good too.

regards,
Rohit.

closed 2015-08-07 by rgrover

leibin2014, 2015-08-07:
Thanks rgrover!