ARMmbed/ble-nrf51822 #46: Cross effect when central and device and unexpected disconnections

fabiencomte, 2015-08-27:
This code to a test triangle with 3 boards.

link 1: board A is central, board B is device
link 2: board B is central, board C is device
link 3: board C is central, board A is device

every time a char is received on serial port of a board, it's reported to the serial ports of the 2 other boards connected.

It works not so bad but depending on connections order, some half links are not working.

For the boards that did first connection as central and second as device, both TX are working.
For the boards that did first connection as device and second as central, half TX are working.

After 20 to 30 s, unexpected disconnections are reported.

#include "mbed.h"
#include "BLE.h"
#include "UARTService.h"
#include "ble/DiscoveredCharacteristic.h"
#include "ble/DiscoveredService.h"
#include "UARTService.h"

#define SOFT_DEVICE_FATHER_HANDLE 3

#define BOARDS_COUNT 3

const Gap::Address_t mac_board_0 = {0xb8, 0xac, 0x4e, 0x8d, 0x8b, 0xeb}; 
const Gap::Address_t mac_board_1 = {0x9c, 0x43, 0x62, 0x30, 0xaf, 0xd2}; 
const Gap::Address_t mac_board_2 = {0x5f, 0x1a, 0x9e, 0x6a, 0x63, 0xdd}; 

// tiny ble board
#define LED_GREEN   p21
#define LED_RED     p22
#define LED_BLUE    p23
#define BUTTON_PIN  p17
#define BATTERY_PIN p1

#define MPU6050_SDA p12
#define MPU6050_SCL p13

#define UART_TX     p9
#define UART_RX     p11
#define UART_CTS    p8
#define UART_RTS    p10

DigitalOut led(LED_RED);
DigitalOut alivenessLED(LED_GREEN);
InterruptIn button(BUTTON_PIN);
AnalogIn battery(BATTERY_PIN);
Serial pc(UART_TX, UART_RX);

bool mac_equals(const Gap::Address_t mac_1, const Gap::Address_t mac_2)
{
    #if 0
        if (mac_1[0] != mac_2[0])
        {
            return false;
        }
        if (mac_1[1] != mac_2[1])
        {
            return false;
        }
        if (mac_1[2] != mac_2[2])
        {
            return false;
        }
        if (mac_1[3] != mac_2[3])
        {
            return false;
        }
[... 454 lines cut ...]
                        else
                        {
                            pc.printf("\r\ncode %d (to device)\r\n", ret);
                        }
                        //ble.waitForEvent();
                    }
                }
                else
                {
                    pc.printf("\r\nnot connected with device\r\n");
                }
            }
           /* 
            while (gatt_server_is_busy)
            {
                
            }
             */   
             
            
        }
        
        ble.waitForEvent(); // save power
    }
}

fabiencomte, 2015-08-31:
please see this post.

https://devzone.nordicsemi.com/question/49705/s130-potential-unstability-case/

I'm not totally sure that is due to S130. Does anyone have an idea ?

rgrover, 2015-09-01:
thanks for pointing us to this discussion. I'm afraid your application is a bit too large for me to review it carefully. Raise this again if no resolution comes about in a few days and I'll try to review.

fabiencomte, 2015-09-01:
I rewritten a pure nordic sdk test application and it works... So there s130 looks correct and the bug is probably in mbed lib side.

rgrover, 2015-09-01:
ok. very good. can you please try replacing Nordic SDK APIs with BLE_API in small steps and see where the failure arises?

fabiencomte, 2015-09-02:
Hello, i'm late on my project so i will try to do that in... few weeks. I'm moving to Nordic SDK to be on schedule.

rgrover, 2015-09-02:
That's a pity. The BLE_API is a thin wrapper around the Nordic SDK and should not result in loss of capability. I hope you find time to uncover the particular issue with BLE_API (if any) which prevented you from developing your application--that would be a very helpful contribution to the rest of the community. We look forward to supporting you in your development.

rainierwolfcastle, 2015-10-15:
http://jira-dev0.cambridge.arm.com/browse/IOTSFW-850

rainierwolfcastle, 2015-10-16:
ARM Internal Ref: IOTSFW-1026