ARMmbed/ble-nrf51822 #50: target dependencies in module.json

jrobeson, 2015-10-11:
Shoudn't it depend on nrf51822 targets like nordic-nrf51822-16k-gcc ? However, those targets are listed in the yotta registry, but they don't actually exist if you try to install them with yotta.  The repositories are invalid

jrobeson, 2015-10-12:
as an aside, do you happen to know where the source code for those nordic targets are hosted? There are some issues in the module.json I'd like to fix. and maybe try to fix up the softdevice logic for s110

rgrover, 2015-10-12:
ble-nrf51822 does get pulled in as a target dependency when using Nordic targets. Nordic targets for mbed OS are still private because we're blocked on Nordic's approval to ship their softdevice and SDK sources as a part of mbed OS. These targets and modules have been ready to be shipped for more than 6 weeks now; but the process is stuck in legal.

jrobeson, 2015-10-12:
aha! thanks for the info

rainierwolfcastle, 2015-10-15:
http://jira-dev0.cambridge.arm.com/browse/IOTSFW-849

rainierwolfcastle, 2015-10-16:
ARM Internal Ref: IOTSFW-1025

jrobeson, 2015-10-24:
looks like this is really close to being completed now! I was able to set a nordic target to get the mbed-drivers and everything. Thanks to everybody at ARM for making this happen.

rgrover, 2015-10-27:
it seems this is not an issue anymore. Please reopen if necessary.

closed 2015-10-27 by rgrover