ARMmbed/ble-nrf51822 #52: Cannot open source input file "system_nrf51.h"

georgeedwards, 2015-10-15:
I have just imported this library into a project. I have set the platform to NRF51-DK (default), but it is throwing the following error:

Error: Cannot open source input file "system_nrf51.h": No such file or directory in "nRF51822/source/nordic-sdk/components/drivers_nrf/hal/nrf51.h", Line: 102, Col: 131

Have I done something wrong, or is this a bug? Are there any work arounds etc. ?

jrobeson, 2015-10-16:
you still have to include the relevant directories from the mbed library. In this case

mbed/libraries/mbed/targets/cmsis/TARGET_NORDIC/TARGET_MCU_NRF51822/system_nrf51.c

rgrover, 2015-10-16:
Can you compare your project against something standard like https://developer.mbed.org/teams/Bluetooth-Low-Energy/code/BLE_iBeacon/ ?

jrobeson, 2015-10-16:
@rgrover : i was running into tons of stuff like this when setting up CMake for mbed projects based on your notebooks (on developer.mbed.org) because everything in this repository, mbed, and the ble libraries had changed paths.

rgrover, 2015-10-16:
as the project evolves, my notes and sample CMakeLists.txt get obsolete in minor ways; but developers would hopefully be able to adapt. I'm happy to provide updates if requested or necessary.

jrobeson, 2015-10-16:
I don't need it for myself, I'm just suggesting that one could easily run into issues like this because of it.

georgeedwards, 2015-10-16:
@rgrover @jrobeson I am using the standard code in this project - https://developer.mbed.org/teams/Bluetooth-Low-Energy/code/BLE_GATT_Example/

rainierwolfcastle, 2015-10-16:
ARM Internal Ref: IOTSFW-1024

georgeedwards, 2015-10-20:
Is there any way I can track the above internal bug? Or are there any know work arounds to this?

rgrover, 2015-10-20:
@georgeedwards We'll mirror parts of our bug database to github. 
Can you please confirm that https://developer.mbed.org/teams/Bluetooth-Low-Energy/code/BLE_iBeacon/ works for you from the online IDE?

georgeedwards, 2015-10-20:
@rgrover  Great, thanks. Yes, the iBeacon example does work for me, it is the GATT Example I am having issues with.

andresag01, 2015-10-21:
@georgeedwards: We had a look at the GATT Example. The main problem was that the APIs used were slightly out-of-date. So, they were updated to the latest version and modified the main.cpp file to comply with the latest APIs. The changes are public, please have a look and let us know if you experience any further issues.
It is important to highlight that the GATT Example worked correctly on the NRF51-DK after a firmware update to version 0218 found under https://developer.mbed.org/platforms/Nordic-nRF51-DK/.

rgrover, 2015-10-21:
@andresag01 :+1:

jrobeson, 2015-12-18:
can this be closed now?

andresag01, 2015-12-18:
The GATT examples was updated.

closed 2015-12-18 by andresag01