ARMmbed/ble-nrf51822 #56: Terrible handling of initLen / minLen and variable length characteristics.

Timmmm, 2015-10-30:
(This is a duplicate of https://github.com/ARMmbed/ble/issues/93  as it affects both projects.)

See: https://devzone.nordicsemi.com/question/34390/no-ble_gatt_hvx_notification-under-20-bytes/?answer=55637#post-id-55637

Basically there are two problems:

1.    The GattCharacteristic initLen parameter is randomly named minLen too. If you follow the code all the way through it actually does two things - it is the initial length (not minimum), and it determines whether the characteristic is variable length or not.

2.    The characteristic for nRF51822 code is only variable length if initLen != maxLen. This is really stupid because you might want the initial length to equal the maximum length but still have variable length. It also isn't documented anywhere.

I suggest the following solution:

1.    Add extra constructors with an extra parameter: bool variableLength.
2.    Name initLen to be initLen everywhere consistently (and not minLen).
3.    Deprecate the old constructors.

rainierwolfcastle, 2015-11-02:
ARM Internal Ref: IOTSFW-1156

andresag01, 2015-12-18:
I believe this has been solved in https://github.com/ARMmbed/ble-nrf51822/commit/3fb32e167fa9c39f31d3786acd0c59485c621e87

closed 2015-12-18 by andresag01