ARMmbed/ble-nrf51822 #57: SoftDevice and SDK documentation

metc, 2015-11-01:
Hello,

I am developing several BLE projects using the nRF51 chip (with home made hardware). Since today, I always used my own C++ libraries / wrapper around the Nordic's SDK.
This library seams to be very interesting and give a very convenient C++ API with a great abstraction to work with BLE on nRF51.

Before starting using your library, I have a few questions to ask (because I did not found much documentation):
  1. The project includes `nRF5xGattClient.h` and `nRF5xGattServer.h`. This means that the library supports both `S110` and `S120` SoftDevices ?
  2. The Nordic SDK is included in the source files, why this ? On which SDK version is this project based on ? Are some sources files of the SDK modified for this project ?

Thanks in advance for these information.

jrobeson, 2015-11-01:
it supports S110 and S130.  I think the sdk files are are from the 8.x.x SDK, but the maintainer can correct me if I'm wrong on that point.

rgrover, 2015-11-02:
This library (ble-nrf51822) is meant to be the adaptation layer between mbed OS's BLE_API and the Nordic SDK/softdevice. Documentation for the use of BLE_API can be found in many places, such as  https://github.com/armmbed/ble, https://docs.mbed.com/docs/ble-intros/en/latest/, and https://developer.mbed.org/teams/Bluetooth-Low-Energy/code/.

We're currently using the S130 softdevice, which includes support for both gatt client and gatt server. We'll soon have a config option of switching to the S110 for those users who would benefit from a little extra SRAM for their application.

We currently use v8.x.x of the Nordic SDK.

closed 2015-11-02 by rgrover

metc, 2015-11-02:
Thank you for our responses and links to the doc.
@rgrover Does the `nordic-sdk/components` contains modified files or are they the official Nordic SDK v8 files ?

The option you mention to switch to S110 is useful (especial for peripheral code running on the 16k RAM chip). Will also the S120 (for central multi-link) be supported in the future ? :blush:

S132 (for nRF52) is API compatible with S130, also looking for the nRF52 support in the future :notes:

rgrover, 2015-11-03:
@metc: files under 'nordic-sdk/components' are fetched from the Nordic SDK. In some cases, minor changes have been made to allow working with our C++ sources; in other rare cases, there may be white-space diffs. Otherwise, the sources are taken directly from Nordic.

metc, 2015-11-03:
Thank you for your answer and support.