ARMmbed/ble-nrf51822 #59: nRF5xn::init don't verify if errors have occurred during btle_init

pan-, 2015-11-03:
See this at: https://github.com/ARMmbed/ble-nrf51822/blob/master/source/nRF5xn.cpp#L94 .
The init process should check and report if something has gone wrong

rainierwolfcastle, 2015-11-04:
ARM Internal Ref: IOTSFW-1168

andresag01, 2015-12-18:
@pan- @rgrover : It is easy to simply return `BLE_ERROR_UNSPECIFIED` when a call to `ble_init()` fails. However, this might not be the ideal solution. The main problem is that we do not have sufficient error codes to be more specific about the cause of the problem. Is it acceptable to return `BLE_ERROR_UNSPECIFIED` when there is a failed call to `ble_init()`? Should we are more error codes to [the BLE API](https://github.com/ARMmbed/ble/blob/develop/ble/blecommon.h#L53)?

pan-, 2015-12-18:
What would you want to return ?

andresag01, 2015-12-18:
Submitted a pull request https://github.com/ARMmbed/ble/pull/155 to add another error code to the API

closed 2015-12-21 by andresag01