ARMmbed/ble-nrf51822 #60: use of extraIncludes

autopulated, 2015-11-05:
This module makes extensive use of `extraIncludes` in module.json (a facility used to add directories to the include path for users of this module). This is bad for two reasons:

 1. these directories contain header files with many names which may clash with any other modules that also rely on including headers without the full path
 2. it substantially increases the length of compilation command lines for any users of this module: there is a finite limit to how long these can be, and at some point could cause problems (especially on windows).

@LiyouZhou just reported an interesting example of the sort of hard-to-debug fault that 1. above causes: Compiling https://github.com/ARMmbed/ble-examples/tree/master/BLE_Thermometer would fail because `#include "BLE.h"` was failing – but only on Linux (causing the problem to go undetected by developers using windows/mac).

The `#include "BLE.h"` statement was in a header file provided by the [ble](https://github.com/ARMmbed/ble) module, where the intention was to include [`ble/BLE.h`](https://github.com/ARMmbed/ble/blob/master/ble/BLE.h). In fact, on windows and mac, [`ble.h`](ble-nrf51822/source/nordic-sdk/components/softdevice/s130/include/ble.h) from this ble-nrf51822 module was being included, which has the full path of `source/nordic-sdk/components/softdevice/s130/include/ble.h`.

Since a previously included header had already included the correct `ble/BLE.h` header, this happened to be harmless, and compilation succeeded (but it's possible this could have caused other weird bugs), breaking only on Linux, where a case-sensitive file system meant `ble.h` was not picked up in place of `BLE.h`.

rgrover, 2015-11-05:
yes, I see the problem with using extraIncludes; but these sources are derived mainly from the Nordic SDK, and if we are to remove extraIncludes, it will mean that we will need to change include paths in many SDK files. That will mean deviating from the standard Nordic SDK; and would be a maintenance headache.

External code will use extraIncludes as a quick way to port to mbed OS. It is unfortunate that there is a ble.h in ble-nrf51822.

rainierwolfcastle, 2015-11-05:
ARM Internal Ref: IOTSFW-1172

pan-, 2016-07-15:
It is fixed since this module does not embed the Nordic SDK anymore.
I close this.

closed 2016-07-15 by pan-