ARMmbed/ble-nrf51822 #76: Memory allocation issue on the NRF51DK board.

marcuschangarm, 2015-11-20:
After upgrading from v0.4.8 to v2.0.7 I'm getting hard faults when using GattClient.

I can circumvent the problem by reverting these two functions from:

https://github.com/ARMmbed/ble-nrf51822/blob/master/source/nRF5xGattClient.cpp#L19-L26
https://github.com/ARMmbed/ble-nrf51822/blob/master/source/nRF5xSecurityManager.cpp#L19-L25

To:

```
nRF5xGattClient &nRF5xGattClient::getInstance(void) {
    static nRF5xGattClient nRFGattClientSingleton;
    return nRFGattClientSingleton;
}

nRF5xSecurityManager &nRF5xSecurityManager::getInstance(void) {
    static nRF5xSecurityManager m_instance;
    return m_instance;
}
```

The test application used: https://github.com/ARMmbed/ble-blocktransfer-mbed

rgrover, 2015-11-23:
@marcuschangarm are the getInstance methoda for GattClient or SecurityManager getting called from an interrupt context? Does the fault happen on a central or a peripheral? Can you provide more information about the context in which the fault happens? Have you got a stack trace?

I'm not too tied to the current implementation using new. I'm also happy with a yotta-config based elimination of GattClient (or even GattServer features).

@andresag01 Could you please work with @marcuschangarm to reproduce/investigate this issue?

marcuschangarm, 2015-11-23:
All my calls are through minar. The problem seems to surface in the gatt client event handler, but this feels more like a memory corruption problem since the location changes when I change the application.

```
#0  HardFault_Handler () at /Users/marcha01/mbed/sdk/ble-blocktransfer-mbed/yotta_modules/mbed-hal-nrf51822-mcu/bootstrap_gcc/startup_nRF51822.S:164
#1  <signal handler called>
#2  0x0411ffde in ?? ()
#3  0x00026332 in processHVXEvent (this=<optimized out>, params=<optimized out>)
    at /Users/marcha01/mbed/sdk/ble-blocktransfer-mbed/yotta_modules/ble/ble/GattClient.h:308
#4  bleGattcEventHandler (p_ble_evt=p_ble_evt@entry=0x20002c64 <btle_init::BLE_EVT_BUFFER>)
    at /Users/marcha01/mbed/sdk/ble-blocktransfer-mbed/yotta_modules/ble-nrf51822/source/btle/btle_discovery.cpp:91
#5  0x0001dfb0 in btle_handler (p_ble_evt=0x20002c64 <btle_init::BLE_EVT_BUFFER>)
    at /Users/marcha01/mbed/sdk/ble-blocktransfer-mbed/yotta_modules/ble-nrf51822/source/btle/btle.cpp:138
#6  0x0001e426 in intern_softdevice_events_execute ()
    at /Users/marcha01/mbed/sdk/ble-blocktransfer-mbed/yotta_modules/ble-nrf51822/source/nordic-sdk/components/softdevice/common/softdevice_handler/softdevice_handler.c:149
#7  0x000248fc in mbed::util::FunctionPointer0<void>::staticcaller (object=<optimized out>, member=<optimized out>, arg=<optimized out>)
    at /Users/marcha01/mbed/sdk/ble-blocktransfer-mbed/yotta_modules/core-util/core-util/FunctionPointer.h:113
#8  0x00020d0e in call (arg=<optimized out>, this=<optimized out>)
    at /Users/marcha01/mbed/sdk/ble-blocktransfer-mbed/yotta_modules/core-util/core-util/FunctionPointerBase.h:80
#9  call (this=<optimized out>) at /Users/marcha01/mbed/sdk/ble-blocktransfer-mbed/yotta_modules/core-util/core-util/FunctionPointerBind.h:44
#10 operator() (this=<optimized out>) at /Users/marcha01/mbed/sdk/ble-blocktransfer-mbed/yotta_modules/core-util/core-util/FunctionPointerBind.h:104
#11 minar::SchedulerData::start (this=0x20003670) at /Users/marcha01/mbed/sdk/ble-blocktransfer-mbed/yotta_modules/minar/source/minar.cpp:471
#12 0x00020d3c in minar::Scheduler::start () at /Users/marcha01/mbed/sdk/ble-blocktransfer-mbed/yotta_modules/minar/source/minar.cpp:296
#13 0x00020874 in main () at /Users/marcha01/mbed/sdk/ble-blocktransfer-mbed/yotta_modules/mbed-drivers/source/retarget.cpp:458
(gdb) f 4
#4  bleGattcEventHandler (p_ble_evt=p_ble_evt@entry=0x20002c64 <btle_init::BLE_EVT_BUFFER>)
    at /Users/marcha01/mbed/sdk/ble-blocktransfer-mbed/yotta_modules/ble-nrf51822/source/btle/btle_discovery.cpp:91
91	                nRF5xGattClient::getInstance().processHVXEvent(&params);
(gdb) list
86	                params.handle     = p_ble_evt->evt.gattc_evt.params.hvx.handle;
87	                params.type       = static_cast<HVXType_t>(p_ble_evt->evt.gattc_evt.params.hvx.type);
88	                params.len        = p_ble_evt->evt.gattc_evt.params.hvx.len;
89	                params.data       = p_ble_evt->evt.gattc_evt.params.hvx.data;
90	
91	                nRF5xGattClient::getInstance().processHVXEvent(&params);
92	            }
93	            break;
94	    }
95	
```

andresag01, 2015-12-18:
It seems that [this commit](https://github.com/ARMmbed/ble-nrf51822/commit/7bf81e7edbf94ffa6f3b06c15299c8e6625c7dee) fixes the issue.

closed 2015-12-18 by andresag01