ARMmbed/ble-nrf51822 #84: setDeviceName() fails to set names longer than 32 bytes

andresag01, 2015-12-03:
It seems that the setDeviceName function call fails in nrf51dk-gcc when supplying a string that is 32 bytes in size or longer. The error returned is `BLE_ERROR_PARAM_OUT_OF_RANGE`. The Bluetooth v4.2 specification in [Vol3 Part C] Section 12.1 states the following:
```
The Device Name characteristic value shall be 0 to 248 octets in length. A device shall have only one instance of the Device Name characteristic.
```

rgrover, 2015-12-04:
The Nordic SDK prevents names larger than 31 bytes to be set. Refer to: https://github.com/ARMmbed/nrf51-sdk/blob/master/source/nordic_sdk/components/softdevice/s130/headers/ble_gap.h#L343
This is a restriction imposed by the partner's stack; they choose to not follow the spec in this case. There isn't much we can do about it.

pan-, 2016-07-15:
This is a limitation from the Nordic stack, there is nothing this abstraction can do about it.
I close this issue.

closed 2016-07-15 by pan-