ARMmbed/ble-nrf51822 #85: A call to shutdown does not clear the state of some components of BLE API

andresag01, 2015-12-08:
The implementation of [`shutdown()`](https://github.com/ARMmbed/ble-nrf51822/blob/master/source/nRF5xn.cpp#L107) does not reinitialise the `GattServer`. For instance, it does not reset the values of characteristicCount, serviceCount, descriptorCount, etc.

rgrover, 2015-12-08:
thanks for raising this. It is a serious flaw. It could be affecting the functioning of Eddystone and similar services which rely upon being able to flush the GattServer based on a shutdown.

andresag01, 2015-12-08:
@rgrover: I believe there is more state that is not being handled correctly in `shutdown()`. For instance, if you have two targets, the first running as a GattServer and the second as a GattClient. If you connect the client to the server and then call `shutdown()` in both this will not clear the Gap state. In other words, if after the call to `shutdown()` I call `getState()` in Gap, then the client will tell me that it is currently "connected" and the server tells me it is "advertising" and "connected".

Clearly, this leaves the API into all sorts of inconsistency and further calls to other functions (even after calling `init()`) might start returning unexpected errors.

rgrover, 2015-12-09:
maybe BLE::shutdown() should call Gap::shutdown(), GattServer::shutdown(), etc. 
We might need to create these additional shutdown virtual methods and leave them empty by default.
Thanks for pointing this out.

closed 2015-12-18 by andresag01