ARMmbed/ble-nrf51822 #133: Compiling crashes on BLEProtocol::AddressType_

chrizmo, 2016-08-16:
When trying to compile a project using ble-nrf51822m the compiler fails with the following output:

>/microbit/micropython/yotta_modules/ble-nrf51822/source/nRF5xGap.cpp: In member function 'virtual ble_error_t nRF5xGap::setWhitelist(const Gap::Whitelist_t&)':
/microbit/micropython/yotta_modules/ble-nrf51822/source/nRF5xGap.cpp:689:59: error: 'BLEProtocol::AddressType_t' is not a class or namespace
if (whitelistIn.addresses[i].type == BLEProtocol::AddressType_t::RANDOM_PRIVATE_NON_RESOLVABLE) {
^
/microbit/micropython/yotta_modules/ble-nrf51822/source/nRF5xGap.cpp: In member function 'ble_error_t nRF5xGap::generateStackWhitelist(ble_gap_whitelist_t&)':
/microbit/micropython/yotta_modules/ble-nrf51822/source/nRF5xGap.cpp:906:61: error: 'BLEProtocol::AddressType_t' is not a class or namespace
if (whitelistAddresses[i].addr_type == BLEProtocol::AddressType_t::RANDOM_PRIVATE_RESOLVABLE) {
^

According to the project in question, it's an upstream problem and a compatibility problem with the GNU toolchain version. The theory is that older versions of gcc does not support the BLEProtocol::AddressType_t  declaration. 

I've tried compiling with gcc 5.4.0/gcc-arm-none-eabi 4.9.3 and gcc 6.1.1/gcc-arm-none-eabi 5-2016-q2-update.

pan-, 2016-08-17:
Could you provide the version of each module you use ? 
You can obtain them by typing `yotta list` at the root of your application.

chrizmo, 2016-08-17:
>microbit-micropython 0.0.1
┣━ mbed-classic 0.0.4
┗━ microbit-dal 1.4.19
  ┣━ ble 2.5.0 yotta_modules/ble
  ┣━ ble-nrf51822 2.5.0 yotta_modules/ble-nrf51822
  ┗━ nrf51-sdk 2.2.0 yotta_modules/nrf51-sdk

dpgeorge, 2016-10-10:
Any progress on this?

pan-, 2016-10-10:
@dpgeorge None, I wasn't able to reproduce the issue and mbed does not support GCC 5.x or 6.X.
If you have any pointers about this issue, that would help.

dpgeorge, 2016-10-10:
We have a handful of users experiencing this problem (at least, those who have reported it); see https://github.com/bbcmicrobit/micropython/issues/338 .  In particular my comment:

> I think the problem is that the "old" gcc version you use can't handle the advanced C++ feature that's used by the code (BLEProtocol::AddressType_t is a typedef of an enum within a struct within a namespace).

pan-, 2016-10-10:
@dpgeorge Thank you, I have a better understanding of the situation now. 
This issue has been fixed a long time ago in this repo: https://github.com/ARMmbed/ble-nrf51822/commit/a6f90e550f93f476aa363ac1be0e29f9d7cf6b81#diff-b69939ef857d8984f2b3197b8a50d152

But it has not been backported in the fork of BLE from the lancaster university: 

```
yt list -i

     ble 2.5.0 (lancaster-university/BLE_API#v2.5.0+mb3) yotta_modules\ble
     ble-nrf51822 2.5.0 (lancaster-university/nRF51822#v2.5.0+mb5) yotta_modules\ble-nrf51822
     nrf51-sdk 2.2.0 (lancaster-university/nrf51-sdk#v2.2.0+mb3) yotta_modules\nrf51-sdk
```

dpgeorge, 2016-10-10:
@pan- yes you're right, it has already been fixed in this repo.  Thanks for pointing it out, now I know better how to fix it for our project.

As far as I'm concerned this issue can be closed.

closed 2016-10-10 by pan-